eclipse-kuksa/kuksa-databroker #6: 0.4.4. Release binariy archives broken

SebastianSchildt, 2024-04-04:
The release archoves in the 0.4.4 release are broken https://github.com/eclipse-kuksa/kuksa-databroker/releases/tag/0.4.4

They are for some reasons "tar.gz" inside of zip, but the bigger issue is, contents is wrong, it misses third party infos and has a dist folder inside.

This is how it looks in current release, i.e.

```
scs2rng@RNG-C-001JT Downloads % ls databroker-arm64.tar.gz.zip
databroker-arm64.tar.gz.zip
scs2rng@RNG-C-001JT Downloads % unzip databroker-arm64.tar.gz.zip 
Archive:  databroker-arm64.tar.gz.zip
  inflating: databroker-arm64.tar.gz  
scs2rng@RNG-C-001JT Downloads % tar -ztvf databroker-arm64.tar.gz 
drwxr-xr-x  0 runner docker      0 Apr  3 21:02 dist/
drwxr-xr-x  0 runner docker      0 Apr  3 21:02 dist/arm64/
-rwxr-xr-x  0 runner docker 4449544 Apr  3 21:02 dist/arm64/databroker
scs2rng@RNG-C-001JT Downloads % 
```

Correct would be

https://github.com/eclipse/kuksa.val/releases/download/0.4.1/databroker-arm64.zip

containing 

```
scs2rng@RNG-C-001JT Downloads % unzip databroker-arm64.zip 
Archive:  databroker-arm64.zip
scs2rng@RNG-C-001JT Downloads % tar -ztvf databroker-arm64.tar.gz                                                     
drwxr-xr-x  0 kuksa  kuksa       0 Oct 30 21:25 ./
drwxr-xr-x  0 kuksa  kuksa       0 Oct 30 21:25 ./databroker/
drwxr-xr-x  0 kuksa  kuksa       0 Oct 30 21:25 ./databroker/thirdparty/
-rw-r--r--  0 kuksa  kuksa       0 Oct 30 21:19 ./databroker/thirdparty/BSD-3-Clause.txt
-rw-r--r--  0 kuksa  kuksa   11356 Oct 30 21:19 ./databroker/thirdparty/Apache-2.0.txt
-rw-r--r--  0 kuksa  kuksa    1337 Oct 30 21:19 ./databroker/thirdparty/BSL-1.0.txt
-rw-r--r--  0 kuksa  kuksa    1552 Oct 30 21:19 ./databroker/thirdparty/BlueOak-1.0.0.md
-rw-r--r--  0 kuksa  kuksa    1210 Oct 30 21:19 ./databroker/thirdparty/Unlicense.txt
-rw-r--r--  0 kuksa  kuksa    2314 Oct 30 21:19 ./databroker/thirdparty/Unicode-DFS-2016.txt
-rw-r--r--  0 kuksa  kuksa    1068 Oct 30 21:19 ./databroker/thirdparty/MIT.txt
-rw-r--r--  0 kuksa  kuksa   70116 Oct 30 21:19 ./databroker/thirdparty/thirdparty_components.txt
-rw-r--r--  0 kuksa  kuksa     709 Oct 30 21:19 ./databroker/thirdparty/ISC.txt
-rw-r--r--  0 kuksa  kuksa   10125 Oct 30 21:19 ./databroker/thirdparty/ring.LICENSE.txt
-rwxr-xr-x  0 kuksa  kuksa 4859160 Oct 30 21:25 ./databroker/databroker
```

erikbosch, 2024-04-04:
The changes seems quite intentional, I will check with @argerus on the background.

erikbosch, 2024-04-08:
0.4.4. release removed. After  long discussion with Johan and Sebastian a proposal for fix exists in https://github.com/eclipse-kuksa/kuksa-databroker/pull/10

erikbosch, 2024-04-15:
PR merged, closing this one

closed 2024-04-15 by erikbosch