eclipse-kuksa/kuksa-databroker #11: Codecov handling

erikbosch, 2024-04-08:
We use [codecov](https://github.com/codecov/codecov-action), see https://github.com/eclipse-kuksa/kuksa-databroker/blob/main/.github/workflows/kuksa_databroker_build.yml

Some problems observed:

```
      - name: Upload coverage to Codecov
        # To use v4 a token must be specifed by "token: ${{ secrets.CODECOV_TOKEN }}""
        # Uploaded result available at https://app.codecov.io/gh/eclipse-kuksa/kuksa-databroker
        uses: codecov/codecov-action@v3
        with:
          files: lcov.info
```

- We use an old version, so we get warnings on Node.js
- By using the old version upload often fail, a known problem, see https://community.codecov.com/t/upload-issues-unable-to-locate-build-via-github-actions-api/3954
- To use the new version we need a token in the eclipse-kuksa orga setup.
- We also do not fail if upload fails, possible we should have settings like in [old repo](https://github.com/eclipse/kuksa.val/blob/master/.github/workflows/kuksa_val_unittest.yml)
```

      uses: codecov/codecov-action@v3
      with:
        files: ${{github.workspace}}/out/all_targets.info
        flags: unittests # optional
        name: codecov-umbrella # optional
        fail_ci_if_error: true # optional (default = false)
        verbose: true # optional (default = false)
```

SebastianSchildt, 2024-05-11:
I am not sure how Eclipse stance on Codecov is, but we are likely not the first project asking for this.

Someone taking this should make a ticket asking for a orga token here https://gitlab.eclipse.org/eclipsefdn/helpdesk/-/issues

(after creating a ticket ping me or @bs-jokri so we could comment with a "virtual" project-lead thumbs up)

erikbosch, 2024-05-13:
There have been projects before

https://gitlab.eclipse.org/eclipsefdn/helpdesk/-/issues/4546

I actually do not know what Otterdog supports here, if you just can request/initiate it by adding it to https://github.com/eclipse-kuksa/.eclipsefdn/blob/main/otterdog/eclipse-kuksa.jsonnet

erikbosch, 2024-05-14:
Upgrade to new version fixed by #31 

Upload worked for latest main, see https://app.codecov.io/github/eclipse-kuksa/kuksa-databroker/commits

But still we could discuss  bit more how we want to track code coverage. With the current setup we cannot expect codecov to succeed for all pull requests (as it depends on that the source fork has a token or has luck). But even if it succeeds uploading - what do we want to happen if coverage is shrinking. Shall that be acceptable?

Leaving this issue open for now

erikbosch, 2024-05-30:
Think this one can be closed for now

closed 2024-05-30 by erikbosch