eclipse-kuksa/kuksa-databroker #23: How to get data from kuksa-databroker using dart gRpc

Athvaith2828, 2024-04-30:
I can able to publish and subscribe to kuksa VSS ( example, Vehicle.speed), using python sdk (kuksa-client), I needs to do the same using gRpc in flutter-dart, I am looking for any basic example for the same . but wouldn't able to find and also would like to know, is it under development as i saw the similar open ticket:  Databroker Api: Use cases + Errors #17

SebastianSchildt, 2024-04-30:
Hi,

While it should be "technically possible", currently, as far as I know, in KUKSA core team we have nobody working on dart and/or flutter.

I know that Automotive Grade Linux guys are into flutter a bit, AND they use some Kuksa, but I am not sure they have some GRPC example in Dart lying around, maybe @g-scott-murray knows and has some pointers to code

g-scott-murray, 2024-05-02:
We have a some examples of using dart-grpc in AGL, the primary ones are:

https://git.automotivelinux.org/apps/flutter-instrument-cluster/tree/lib/vehicle-signals
(see vss_client.dart and associated provider .dart files)

and:

https://git.automotivelinux.org/apps/flutter-ics-homescreen/tree/lib/data/data_providers
(see val_client.dart and associated provider .dart files, notable vehicle_notifier.dart)

I can't say I would consider any of that code particularly great, and we do have some known issues, e.g. we see an issue in the homescreen with subscription failing after running for a while, and I've so far not had a lot of success handling it and getting back into a working state.  That's on my TODO list to look into further in the next couple of months.  I may also start looking at using non-Dart code (e.g. tonic in Rust) via FFI, as Toyota tells us that they're doing similar for quite a few things in their homescreen implementation.

Athvaith2828, 2024-05-03:
Thanks for the quick reply @SebastianSchildt and @g-scott-murray ,
I have tried to connect to kuksa-databroker using dart after generating proto (kuksa/val/v1)
and When I made rpc calls (subscribe and getserverinfo) , I gets the error
**gRPC Error (code: 14, codeName: UNAVAILABLE, message: Error connecting: SocketException: Connection refused (OS Error: Connection refused, errno = 111), address = localhost, port = 38508, details: null, rawResponse: null, trailers: {})**
and this is the databroker docker output
**docker run --rm -it --name Server --network kuksa -v ./certificates:/opt/kuksa ghcr.io/eclipse-kuksa/kuksa-databroker:main --insecure 
2024-05-03T11:29:05.099888Z  INFO databroker: Init logging from RUST_LOG (environment variable not found)
2024-05-03T11:29:05.099924Z  INFO databroker: Starting Kuksa Databroker 0.4.4
2024-05-03T11:29:05.101730Z  INFO databroker: Populating metadata from file 'vss_release_4.0.json'
2024-05-03T11:29:05.372554Z  WARN databroker: Authorization is not enabled.
2024-05-03T11:29:05.372609Z  INFO databroker::broker: Starting housekeeping task
2024-05-03T11:29:05.372626Z  INFO databroker::grpc::server: Listening on 0.0.0.0:55555
2024-05-03T11:29:05.372632Z  INFO databroker::grpc::server: TLS is not enabled
2024-05-03T11:29:05.372635Z  INFO databroker::grpc::server: Authorization is not enabled.**

Is this endpoints are unavailable ?
and In dart , can we not use as @g-scott-murray said ( he is also experiencing disconnections) ?

Athvaith2828, 2024-05-07:
Thanks for the reply and reference @g-scott-murray, 
And one more doubt @SebastianSchildt , I can see the above example uses protos from kuksa/val/v1
Is protos from sdv/databrokerl/v1 is also active in while running ? if yes, then what is the difference and when to use which ?

SebastianSchildt, 2024-05-11:
"sdv/databrokerl/v1" is an older API. In current main builds it needs to be explicitely enabled with `--enable-databroker-v1`,  when using the Python SDK only the newer kuksa/val/v1 API (and protos) are used.

If API changes again, usually the strategy is to have a suitably long grace period where the old API is still available. Currently the recommendation is to use only the kuksa/val/v1 , unless you have an urgent need for very high performance/high frequency setting of values. That is being worked on in a newer API (version) or  - unfortunately limited to current values only - in the sdv1 API.

closed 2024-08-28 by Athvaith2828