eclipse-kuksa/kuksa-databroker #46: Unsatisfied trait bounds do not allow lint-source to succeed with active rustc toolchain 1.79.0

noci0001, 2024-06-19:
While trying to implement kuksa databroker as a crate dependency to .get_current_values() from it, I am not passing the lint-source code test because of unsatisfied trait bounds inside the kuksa-databroker. My assumption is that the current version of the databroker uses rustc toolchain 1.75.0 not is currently not compatible with the active 1.79.0 requested by the project I am contributing to.

Here is the list of errors when trying to build with rustc 1.79.0

```
error[E0277]: the trait bound `Channel: tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::status::Status>>>` is not satisfied
   --> /home/runner/.cargo/git/checkouts/kuksa-databroker-with-exposed_value-961b410d4ef6bea2/8b6d205/lib/kuksa/src/lib.rs:35:13
    |
34  |         let mut client = proto::v1::val_client::ValClient::with_interceptor(
    |                          -------------------------------------------------- required by a bound introduced by this call
35  |             self.basic_client.get_channel().await?.clone(),
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::status::Status>>>` is not implemented for `Channel`
    |
    = help: the trait `tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::Status>>>` is implemented for `Channel`
    = help: for that trait implementation, expected `tonic::Status`, found `tonic::status::Status`
note: required by a bound in `ValClient::<T>::with_interceptor`
   --> /home/runner/work/fleet-management-private/fleet-management-private/components/target/debug/build/databroker-proto-a2c47de32a951d09/out/kuksa.val.v1.rs:730:16
    |
723 |           pub fn with_interceptor<F>(
    |                  ---------------- required by a bound in this associated function
...
730 |               T: tonic::codegen::Service<
    |  ________________^
731 | |                 http::Request<tonic::body::BoxBody>,
732 | |                 Response = http::Response<
733 | |                     <T as tonic::client::GrpcService<tonic::body::BoxBody>>::ResponseBody,
734 | |                 >,
735 | |             >,
    | |_____________^ required by this bound in `ValClient::<T>::with_interceptor`

error[E0277]: the trait bound `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor` is not satisfied
   --> /home/runner/.cargo/git/checkouts/kuksa-databroker-with-exposed_value-961b410d4ef6bea2/8b6d205/lib/kuksa/src/lib.rs:36:13
    |
34  |         let mut client = proto::v1::val_client::ValClient::with_interceptor(
    |                          -------------------------------------------------- required by a bound introduced by this call
35  |             self.basic_client.get_channel().await?.clone(),
36  |             self.basic_client.get_auth_interceptor(),
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `FnMut<(tonic::request::Request<()>,)>` is not implemented for `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_`, which is required by `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor`
    |
    = note: required for `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_` to implement `tonic::service::interceptor::Interceptor`
note: required by a bound in `ValClient::<T>::with_interceptor`
   --> /home/runner/work/fleet-management-private/fleet-management-private/components/target/debug/build/databroker-proto-a2c47de32a951d09/out/kuksa.val.v1.rs:728:16
    |
723 |         pub fn with_interceptor<F>(
    |                ---------------- required by a bound in this associated function
...
728 |             F: tonic::service::Interceptor,
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `ValClient::<T>::with_interceptor`

error[E0599]: the method `set` exists for struct `ValClient<InterceptedService<Channel, impl FnMut(Request<()>) -> Result<Request<()>, Status>>>`, but its trait bounds were not satisfied
  --> /home/runner/.cargo/git/checkouts/kuksa-databroker-with-exposed_value-961b410d4ef6bea2/8b6d205/lib/kuksa/src/lib.rs:44:22
   |
44 |         match client.set(set_request).await {
   |                      ^^^ method cannot be called due to unsatisfied trait bounds
   |
  ::: /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tonic-0.9.2/src/transport/channel/mod.rs:69:1
   |
69 | pub struct Channel {
   | ------------------ doesn't satisfy `_: Service<Request<UnsyncBoxBody<Bytes, Status>>>`
   |
   = note: the full type name has been written to '/home/runner/work/fleet-management-private/fleet-management-private/components/target/debug/deps/kuksa-4374078716ce5e2b.long-type-3359690268834603901.txt'
   = note: consider using `--verbose` to print the full type name to the console
   = note: the following trait bounds were not satisfied:
           `Channel: tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::status::Status>>>`
           `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: FnMut<(tonic::request::Request<()>,)>`
           which is required by `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor`
           `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: FnOnce<(tonic::request::Request<()>,)>`
           which is required by `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor`

error[E0277]: the trait bound `Channel: tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::status::Status>>>` is not satisfied
[... 230 lines cut ...]
    |                ---------------- required by a bound in this associated function
...
728 |             F: tonic::service::Interceptor,
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `ValClient::<T>::with_interceptor`

error[E0599]: the method `subscribe` exists for struct `ValClient<InterceptedService<Channel, impl FnMut(Request<()>) -> Result<Request<()>, Status>>>`, but its trait bounds were not satisfied
   --> /home/runner/.cargo/git/checkouts/kuksa-databroker-with-exposed_value-961b410d4ef6bea2/8b6d205/lib/kuksa/src/lib.rs:351:22
    |
351 |         match client.subscribe(req).await {
    |                      ^^^^^^^^^
    |
   ::: /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tonic-0.9.2/src/transport/channel/mod.rs:69:1
    |
69  | pub struct Channel {
    | ------------------ doesn't satisfy `_: Service<Request<UnsyncBoxBody<Bytes, Status>>>`
    |
    = note: the full type name has been written to '/home/runner/work/fleet-management-private/fleet-management-private/components/target/debug/deps/kuksa-4374078716ce5e2b.long-type-3359690268834603901.txt'
    = note: consider using `--verbose` to print the full type name to the console
    = note: the following trait bounds were not satisfied:
            `Channel: tower_service::Service<http::Request<http_body::combinators::box_body::UnsyncBoxBody<bytes::bytes::Bytes, tonic::status::Status>>>`
            `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: FnMut<(tonic::request::Request<()>,)>`
            which is required by `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor`
            `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: FnOnce<(tonic::request::Request<()>,)>`
            which is required by `impl FnMut(tonic::Request<()>) -> Result<tonic::Request<()>, tonic::Status> + '_: tonic::service::interceptor::Interceptor`
```

noci0001, 2024-06-19:
To reproduce this error, you can do the following:
- inside this project repo, cd lib/kuksa
- rustc install 1.79.0
- rustc override set 1.79.0
- cargo build

erikbosch, 2024-07-02:
Thank you for reporting this - sorry for not detecting this issue earlier. I did some quick investigations. The CI actually use 1.79.0 as I believe that is the latest stable rust version, but it does not seem that we build things in CI the way you describe above. I tried adding a new CI step and succeeded reproducing the error, see:

- https://github.com/erikbosch/kuksa-databroker/blob/main/.github/workflows/kuksa_databroker_build.yml#L73
- https://github.com/erikbosch/kuksa-databroker/actions/runs/9763128595/job/26948270520

I am not really a Rust expert, maybe some of my colleagues @lukasmittag @rafaeling has more insight why we only get an error if building lib/kuksa explicitly.

erikbosch, 2024-07-02:
Some more investigation indicates that it maybe is a tonic mismatch, will likely come with a PR with a solution soon

erikbosch, 2024-07-02:
@noci0001 - feel free to test if upgrading the tonic version like in #48 solves the problem also in your setup.

PR #48 (2024-07-03): Update lib dep and and add build step
This PR changes so we use same tonic version in all Cargo.toml files. It also adds a job to make sure that lib is always buildable.

Fixes #46

RCA:

We already used tonic 0.11.0 in https://github.com/eclipse-kuksa/kuksa-databroker/blob/main/Cargo.toml and it seemed that we did not build "lib" in CI as a separate build, so that we did not detect that the lib Cargo file specified tonic 0.9.1 . When upgrading tonic I could not reproduce the error any longer,
commit aff9e8d4 (2024-07-03): Update lib dep and and add build step

Fixes #46
closed 2024-07-03 by lukasmittag