eclipse-kuksa/kuksa-databroker #57: Using Custom VSS Data Entries - Missing volume mapping in the user_guide.md

PySchilling, 2024-09-04:
The user_guide.md give the following example for loading a custom specification file in the chapter Using Custom VSS Data Entries.

`docker run --rm -it -p 55555:55555 ghcr.io/eclipse-kuksa/kuksa-databroker:main --insecure --vss vss.json`

See: [https://github.com/eclipse-kuksa/kuksa-databroker/blob/main/doc/user_guide.md#using-custom-vss-data-entries](https://github.com/eclipse-kuksa/kuksa-databroker/blob/main/doc/user_guide.md#using-custom-vss-data-entries)

This results in the following error message:

`Error: Os { code: 2, kind: NotFound, message: "No such file or directory" }`

The solution would be to map the file into the docker container.

`docker run --rm -it -p 55555:55555 -v ./vss.json:/vss.json ghcr.io/eclipse-kuksa/kuksa-databroker:main --insecure --vss vss.json`

Maybe this should be added to the documentation.

lukasmittag, 2024-09-04:
If you would be so kind and create a PR for this we would appreciate it :)

SebastianSchildt, 2025-02-06:
@lukasmittag  not sure "wontfix" is a good label here. While we do not want to replace "generic docker docs", the example as given in the linked doc can never work. 

Running with docker you _need_ to mount anything you want to use inside. So eother we add a mount parameter to the example, or we show databroker start without docker

lukasmittag, 2025-02-06:
but it uses the pre loaded file which is newest vss version if you "just start it"

PR #162 (2025-07-07): Fix user guide on how to mount custom vss
Fixes the user guide on how to mount custom VSS when running databroker via docker run.

Fixes #57
closed 2025-07-07 by SebastianSchildt