eclipse-kuksa/kuksa-databroker #66: Custom Data Topics in Kuksa Databroker

lapnd, 2024-09-13:
Hi,

I am interested in understanding the capabilities of Kuksa Databroker regarding custom data topics. Specifically, I would like to know if it is possible to configure Kuksa Databroker to support custom data types, such as images taken from a camera.

Can Kuksa Databroker handle a scenario where an image is published to the broker by one application, and another application subscribes to that image for processing? If so, could you please provide guidance or examples on how to set this up?

Thank you!

SebastianSchildt, 2024-09-17:
Hi, so KUKSA can not support "custom datatypes" (i.e. datatypes not in VSS), you could get away with uing an uint8 array (i.e. a byte array) for binary data, but probably that won't be very efficient, as KUKSA ist not really designed for high bandwidth but more for low latency small data.

In generall in VSS (which as data model is also not really desinged for auio/video) is _refernecing_ such sources with URIs, see e.g. here https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/index.html#data-streams

If you have a real high performance use case (e.g. thinking about somehting like ADAS), there a specific stacks such as https://iceoryx.io

lapnd, 2024-09-18:
Thank you @SebastianSchildt for your information.

closed 2024-09-18 by lapnd