eclipse-kuksa/kuksa-databroker #133: Is

venkatkonjeti-cariad, 2025-02-04:
The SQL syntax documented on this page https://github.com/eclipse-kuksa/kuksa-databroker/blob/main/doc/QUERY.md says it is only relevant for the sdv.databroker.v1. Was it discontinued in the kuksa.val.v1 and above versions? If yes, the alternate is documented as shown in the python sdk here? https://github.com/eclipse-kuksa/kuksa-python-sdk/blob/main/docs/examples/async-grpc.md

Thanks
Venkat

mikehaller, 2025-02-06:
Due to a finding in a security audit, the SQL syntax was permanently disabled for SDV V1. The API is by default disabled, but will be kept for some time onwards. There are no known plans yet to re-add the feature to newer APIs.

Contributions welcome, however we'd like to have a secure design and implementation and generic SQL-like syntaxes are a bit more complicated and take more effort to make them right. Hence, we decided that for production use-cases, it's of lower priority and we can postpone it to future releases.

Do you have a specific use cases in mind which goes beyond the obvious and usual developer convenience (not saying that is unimportant, just that we're aware of it).

venkatkonjeti-cariad, 2025-02-10:
Thanks @mikehaller for the details.

The use case we are trying was, subscribe for telemetry based on filters, so that we can receive telemetry only condition matches. Also, managing these rules through an external file avoids need to deploy the code again. We felt SQL is string so replacing this dynamically going to handy.

Example just for demonstration:
* vehicle.speed > 50
* Fan.speed > 3

Scenarios:
* Update existing signal subscription vehicle.speed > 80
* Adding new signal to subscribe Door.Row3.Left.Window.Position

Adding and updates to subscription wanted to enable without deployments if there is an option.

Please let me know if you have any thoughts on this.