eclipse-kuksa/kuksa-databroker #150: Support for Unix Domain Sockets (UDS) in CLI/SDK & JWT Authorization

dotChris90, 2025-04-01:
Hello community,

I noticed that the data broker supports Unix Domain Sockets (UDS), but I couldn't find a way to connect to it using the CLI or SDK. Is there a supported method for this? I assumed UDS would be faster than TCP/IP.

Additionally, I would like to know if JWT-based authorization is also supported when using UDS.

Thanks in advance!

mikehaller, 2025-04-11:
Currently no support for UDS for the CLI. Yeah, it's faster, so we assume the CLI would not need that -> low priority.

Care to contribute?

dotChris90, 2025-04-25:
After my vacation I will check how much effort it is. ;) currently in vacation so I didn't react. I saw that in your prof tooling you already accept Unix Sockets so that would be a starting point.